nion-software/nionswift-instrumentation-kit #8: Exception loading a camera prevents the whole instrumentation UI from loading

Brow71189, 2019-09-25:
see here for an example traceback:
https://drive.google.com/file/d/1nkT0XeJTwUwjV368lLJ7KDtDs-bNHCUo/view?usp=sharing

cmeyer, 2019-09-25:
When I test this with the simulator, the 2nd camera still loads and has a UI. The camera for which there was an exception (the 1st camera) does not load nor is there a UI. What do you expect to happen if the camera can't be initialized due to an exception? I think the UI is not going to be valid in that case.

Code I used for the simulator:

```
--- a/nionswift_plugin/usim/CameraDevice.py
+++ b/nionswift_plugin/usim/CameraDevice.py
@@ -350,6 +350,8 @@ class CameraSettings:
 
     def get_current_frame_parameters(self) -> CameraFrameParameters:
         """Get the current frame parameters."""
+        if self.__camera_id == "usim_ronchigram_camera":
+            raise Exception("BAD")
         return CameraFrameParameters(self.__frame_parameters)
```

Brow71189, 2019-09-25:
Of course the failing camera is not valid and it is fine if it is not shown.
In the case I mentioned (that was in Vienna) Clemens told me that also the ronchigram camera did not show up because the eels camera had the exception. So looks like we have to try to reproduce it there.

cmeyer, 2019-09-25:
I think it depends on if the camera is loaded before or after the instrumentation kit. If before (Vienna case) then the component is registered in the for loop on line 1177 of `camera_base.py`; but if after, it gets registered using an event, set up on line 1173. In the case of an event, the exception is handled by the event; in the case of the for loop, there is no exception handling. I'll make a patch (it's ready to go, I just couldn't reproduce the problem).

closed 2019-09-25 by cmeyer

Brow71189, 2019-09-25:
Great, thanks.

commit 30e5b1d2 (2019-09-25): Improve exception handling during camera initialization. Fixes #8.