nion-software/nionswift-instrumentation-kit #32: Set "processing" to "None" if "Images" is selected in UI

Brow71189, 2020-06-26:
If "processing" was set to "sum_project" before it will not be changed and the SI tool will always acquire spectra, even thought the user selected "Images" in the UI. 

https://github.com/nion-software/nionswift-instrumentation-kit/blob/f5e65efb4b63effcb14d5ea1e77a88d534f28545/nionswift_plugin/nion_instrumentation_ui/ScanAcquisition.py#L279

cmeyer, 2020-06-26:
This bug title is describing a fix -- but what is the problem? Can you change the title to describe the bug and also describe how to reproduce it?

Brow71189, 2020-07-06:
see nion-software/nion-dectris#16 for more details

cmeyer, 2025-03-20:
@Brow71189 Can I close this issue and make a new issue with the title "Devices should have a way to validate frame parameters and provide feedback to the user?"

Brow71189, 2025-03-21:
Don't we already have that? https://github.com/nion-software/nionswift-instrumentation-kit/blob/dd65f8023286c7776a81f55c3017a384a04be230/nion/instrumentation/camera_base.py#L594

This is the validate function and I think it is working as expected.

cmeyer, 2025-03-21:
Ok, so this issue can be closed?

Brow71189, 2025-03-24:
I think we can close this. At least I've not seen any issues like this anymore lately.

closed 2025-03-24 by Brow71189