nion-software/nionswift-instrumentation-kit #33: Make frame paramters handling more reliable and transparent

Brow71189, 2020-06-26:
Right now, frame paramters can be set in too many different ways and it feels arbitrary when which method is used.

We have:

``set_frame_parameters()``, ``set_current_frame_parameters()``, ``set_record_frame_parameters()``
Also ``set_selected_profile_index()`` can change frame paramters.

It is not clear what side effects these methods have. Some put the frame paramters into the settings list, some only into a temporaray place.
If you get frame paramters into the wrong place once, they get saved like this and will cause problems even after a restart of Swift. See #32 for an example that is a consequence of this system.

I think we should either limit the copies of frame parameters that are floating around to a smaller number, or define very clearly what each of the functions that manipulate them do and in which cases they get called by e.g. the SI tool.

The current situation is extremeley difficult to deal with for camera implementers / maintainers of camera code. One small change either in the camera plugin itself or some other acquisition tool can break acquisition in subtle ways, simply because some frame parameters object went into the wrong place once due to using for example ``get_current_frame_parameters`` in combination with ``set_frame_parameters``.

cmeyer, 2020-06-26:
This definitely needs some work. But these are largely not public and/or deprecated methods.

Can you file issues to request what you're trying to accomplish? (or describe the use cases here)

Brow71189, 2020-07-06:
see nion-software/nion-dectris#16 for more details