nion-software/nionswift-instrumentation-kit #34: Section-wise updates of SI/4D images causes larger maps to hang and not complete

Brow71189, 2020-07-28:
https://github.com/nion-software/nionswift-instrumentation-kit/blob/5541b2e15665b8354f76a63a4e8fbfaf43e2e5f2/nion/instrumentation/scan_base.py#L780

This updates the `camera_data_channel` once per second by default which also writes the whole SI/4D data out to disk. If writing it takes significantly longer than the update period (i.e. 1 s), this causes Swift to hang and the map never finishes (or maybe it finishes after a very long time, but a normally 3 min acquisition with 16 GB of data did not finish after 20 min).

I don't know where this needs to get fixed, but on problem is for sure that here:

https://github.com/nion-software/nionswift-instrumentation-kit/blob/5541b2e15665b8354f76a63a4e8fbfaf43e2e5f2/nionswift_plugin/nion_instrumentation_ui/ScanAcquisition.py#L120

the `item_transaction` gets created but its `__enter__()` method is never called as far as I can tell, so saving to disk is not actually delayed until all the data has been acquired.

However, for now this bug completely prevents acquiring any sort of data that is significantly larger than what can be written to disk within about a second.

(reported by BH from HU)

cmeyer, 2020-08-02:
The data does not seem to be explicitly writing to disk during acquisition - however, since the data is h5py data, it is implicitly written to disk whenever it changes.

Also, entering the transaction in a context manager is a no-op. The only reason to use a context manager is to close the transaction; but this is done on line 148.

In the partial acquisition algorithm for initial development the implementation was to copy the entire section to the destination instead of taking advantage of the partial nature of the data. So I'm making a change that will only copy the new part of the acquisition to the destination. This should greatly reduce memory and [due to h5py memory mapped files] writing to disk.

cmeyer, 2020-08-02:
Can you try this patch? The patch breaks the multi acquire tests (in general, multi acquire needs an overhaul - the threading display queue is excessively complicated - we'll chat about this offline).

```
diff --git a/nion/instrumentation/scan_base.py b/nion/instrumentation/scan_base.py
index a08bca2..c642fe2 100644
--- a/nion/instrumentation/scan_base.py
+++ b/nion/instrumentation/scan_base.py
@@ -752,6 +752,7 @@ class ScanHardwareSource(HardwareSource.HardwareSource):
                             uncropped_xdata = partial_data_info.xdata
                             is_complete = partial_data_info.is_complete
                             is_canceled = partial_data_info.is_canceled
+                            last_valid_rows = 0  # valid rows within section
                             # this loop is awkward because to make it easy to implement synchronized begin in a backwards
                             # compatible manner, it must return all of its data on the first call. this means that we need
                             # to handle the data by sending it to the channel. and this leads to the awkward implementation
@@ -763,16 +764,21 @@ class ScanHardwareSource(HardwareSource.HardwareSource):
                                 metadata["hardware_source"] = copy.deepcopy(scan_info.camera_metadata)
                                 metadata["scan"] = copy.deepcopy(scan_info.scan_metadata)
                                 metadata["instrument"] = copy.deepcopy(scan_info.instrument_metadata)
-                                metadata["scan"]["valid_rows"] = section_rect[0][0] + section_rect[1][0]
+                                metadata["scan"]["valid_rows"] = section_rect[0][0] + partial_data_info.valid_rows
                                 partial_xdata = crop_and_calibrate(uncropped_xdata, flyback_pixels, scan_calibrations, data_calibrations, data_intensity_calibration, metadata)
-                                if camera_data_channel:
+                                if camera_data_channel and partial_data_info.valid_rows > 0:
                                     data_channel_state = "complete" if is_complete and is_last_section else "partial"
                                     data_channel_data_and_metadata = partial_xdata
-                                    data_channel_sub_area = Geometry.IntRect(Geometry.IntPoint(), Geometry.IntSize.make(data_channel_data_and_metadata.collection_dimension_shape))
+                                    # src rect is the area of the section data that will be copied
+                                    partial_size = Geometry.IntSize(height=partial_data_info.valid_rows - last_valid_rows, width=section_rect.width)
+                                    src_rect = Geometry.IntRect(Geometry.IntPoint(y=last_valid_rows), partial_size)
+                                    # dst rect is the destination in the final layout.
+                                    dst_rect = src_rect + section_rect.origin
                                     data_channel_view_id = None
                                     camera_data_channel.update(data_channel_data_and_metadata, data_channel_state,
                                                                Geometry.IntSize(h=scan_param_height, w=scan_param_width),
-                                                               section_rect, data_channel_sub_area, data_channel_view_id)
+                                                               dst_rect, src_rect, data_channel_view_id)
+                                    last_valid_rows = partial_data_info.valid_rows
                                 # break out if we're complete
                                 if is_complete:
                                     break
```

Brow71189, 2020-08-03:
I tried the patch on APM2 and it does indeed get rid of the perma-saving problem.
Howerver, acquiring large maps is still not possible.
If you're lucky, Swift hangs at the beginning and the end of the map during writing the whole dataset to disk.
But most of the time the saving at the end never finishes. Something must get stuck and causes disk usage to stay at about 700 MB/s for forever (I let it sit there for about 45 min and it did not manage to finish).
The parameters I tried were: 4D image with 128x128 scan pixels and 400x2048 pixels on the EELS camera. Dwell time was 10 ms.
This results in a total size of about 50 GB and a line time of 1.28 s with a size of about 400 MB per line.
That should be very easy to directly write to disk on a SSD (which I also used).

I also noticed excessive memory usage: The above map used up the whole 128 GB of memory on the computer causing errors like this occasionally:
```
File "c:\nionapps\developer\master\nionswift\nion\swift\model\DataItem.py", line 1116, in set_data_and_metadata_partial
self.__data_and_metadata.data[dst] = data_and_metadata.data[src]
File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
File "C:\NionApps\Miniconda3\envs\developer\lib\site-packages\h5py\_hl\dataset.py", line 640, in __setitem__
val = numpy.asarray(val, order='C')
File "C:\NionApps\Miniconda3\envs\developer\lib\site-packages\numpy\core\_asarray.py", line 85, in asarray
return array(a, dtype, copy=False, order=order)
MemoryError
:
Unable to allocate 18.0 GiB for an array with shape (46, 128, 400, 2048) and data type float32
```

In the rare case that the acquisition was successful, the 50 GB of the map did also not get freed after the acquisition was finished which definitely causes a subsequent acquisition to fail.

cmeyer, 2020-08-03:
A simplifying assumption for acquisition has always been that the entire acquisition will fit in memory. This can be relaxed by doing section-by-section acquisition, which was developed for drift correction but does not require drift correction to be enabled. Each section will need to be in memory.

Can you try hard coding a value for `section_height = 4` around line 293 of `ScanAcquisition.py`?

Doing this might also highlight the problem with the last partial update too. That sounds like a separate problem that might need debugging.

commit 54ecd9ac (2020-08-04): Fix #34. Ensure partial acquisition only updates new rows.

Also breaks multi-SI test. Disabled for now. Requires
rewrite of multi-SI to use non-thread data item
updates.
closed 2020-08-04 by cmeyer