nion-software/nionswift-instrumentation-kit #40: Consider expanding partial data structure to include valid row range

cmeyer, 2020-08-19:
The paradigm that partial acquisition uses is to allocate the full memory in acquire_synchronized_begin and then acquire sections of that in acquire_synchronized_continue, returning the number of valid rows each time acquire_synchronized_continue returns.

So if you change out the memory, it will still be trying to index into that starting at the last row. For instance, if in the first call to acquire_synchronized_continue, you return memory A and 20 rows complete, it will copy out rows 0:20 from A. Now if you return B and 40 rows complete, it will copy out 20:40 from B. That doesn't work. Instead the idea is to return A and 40 rows complete in which case it will copy out 20:40 from A (which was allocated fully during acquire_synchronized_begin.

I think a better strategy is to change this system so that acquire_synchronized_continue is passed the last acquired row and returns the range of valid rows in the returned memory. The last acquired row is passed so that the camera device doesn't have to track that internally (simplifies the camera device implementation). And the valid rows allows you to allocate new memory each time and return 0:N where N is the number of rows in the partial acquisition. And this plan degenerates to the current behavior if you return LAST:LAST+N and the same memory each time.

Brow71189, 2020-08-20:
I still think that #39 should be merged and that the current implementation has a bug.

The loop in `grab_synchronized` does something like this:
```python
partial_data_info = acquire_synchronized_begin()
xdata = partial_data_info.xdata
is_complete = partial_data_info.is_complete
is_canceled = partial_data_info.is_canceled

while not is_canceled:
    
    valid_rows = partial_data_info.valid_rows
    
    # Do something with the data
    
    if is_complete:
        break
    
    # We get a new "partial_data_info" object and use the updated values for "is_complete" and "is_canceled" ...
    partial_data_info = acquire_synchronized_continue()
    is_complete = partial_data_info.is_complete
    is_canceled = partial_data_info.is_canceled
    
    # ... But we assume that "partial_data_info.xdata" is still referencing the same object than before (not the same memory, the same object
    # wrapping the memory!!!). So why do we even return a new "partial_data_info" object then if we do not fully use the information provided?
    # So if "acquire_synchronized_begin" did not produce any data (i.e. xdata is None), then it will still be None here, even after we called
    # "acquire_synchronized_continue", so the loop will stop.
    # Or if the wrapper object (i.e. the "DataAndMetadata" object) changed, we are referencing the wrong object now
    if not xdata:
        break
```

I think even if it can work in the current implementation, this is definitely a logical error: We are receiving a new `partial_data_info` in each iteration, but we are not using all the information provided by it. We mix the information from an earlier object with the current one and rely on all this information fitting together and still being valid.
If we would instead only use the information contained in one `partial_data_info` object it is justifiable to assume that the information in this one object is self-consitent and valid. But for combining information from multiple calls to a function and multiple objects I don't think it is justifiable that this information is self-consistent and valid.

cmeyer, 2020-08-20:
Yes - the logical error/inconsistency is/was there to support our older camera device implementations that did not implement partial acquisition and returned the full data from acquire_synchronized_begin. We might consider dropping that support since we've decided to sync the releases (nionswift and nionswift-instrumentation-kit) anyway.

And I'm fine making each partial data return value independent of the others (i.e. keeping your merge).

But we still have the problem that, practically, a camera device must still allocate the full array in acquire_synchronized_begin since it cannot predict when the next call to acquire_synchronized_continue will occur and must record all data until the next call, up to the full acquisition. That implies that PartialData will always return the same xdata (the one allocated in acquire_synchronized_begin).

I guess it's possible that a camera device chunks its data into smaller individual arrays; and acquire_synchronized_continue can return those completed segments as they are acquired... is that how you're planning it? If so, I'm not sure that's a good plan. My experience has been that allocating memory on-the-fly during acquisition can slow things down substantially, particularly for cameras that require contiguous physical RAM, but that claim needs more evidence/experimentation. Also, it might not be as big of an issue on modern architectures.

Brow71189, 2020-08-20:
I know that the camera needs to allocate the whole block of memory internally, but that doesn't mean that it needs to return that whole block each time `acquire_synchronized_continue` is called.
What I'm doing specifically is that I return a new `DataAndMetadata` object with each call to `acquire_synchronized_continue` that contains only the slice of the big data block up to `valid_rows`.
The data does not actually get copied but the shape of the numpy array in the xdata object is different every time `acquire_synchronized_continue` is called.

cmeyer, 2020-08-20:
That's almost the same thing then - the problem is that the higher level code which copies it out always copies data from `last_valid_rows:valid_rows`. So if you return a smaller slice of the data, it will try to pull data from the wrong rows.

I think the solution is to implement this issue: return a valid_row_range that can specify a range of `0:valid_rows - last_valid_rows` instead of the implicit range of `last_valid_rows:valid_rows`.