nion-software/nionswift-instrumentation-kit #42: Consider including flyback_pixels in acquire_synchronized functions

Brow71189, 2020-08-27:
e.g. `acquire_synchronized_prepare(scan_shape, flyback_pixels)`

Then the partial data structure should probably also include them, so that the acqusition software knows that the camera removed them already (related to #40).

This would be one way to fix nion-software/nionswift#527.

Brow71189, 2020-08-28:
It looks like the extra copy in h5py is not a major performance killer as long as we have enough free memory to make a copy of the chunks we are writing.
This means that with the partial updates for SI/4D acquisition, we typically do not run into an issue here, even if datasets get close to the size of the memory that is installed in a system.
Just when writing large arrays in one chunk you can run into troubles with the extra copy.
As far as I can tell the copy operation releases the GIL, so it happens in the background and does not affect an ongoing acquisition for example.

cmeyer, 2020-08-28:
I updated the docs for synchronized acquisition functions (see the inline comments in [camera_base](https://github.com/nion-software/nionswift-instrumentation-kit/blob/master/nion/instrumentation/camera_base.py#L359)).

TL;DR: `acquire_synchronized_prepare` is deprecated. But it would be useful to pass some scan parameters such as flyback and perhaps others to `acquire_synchronized_begin`.