nion-software/nionswift-instrumentation-kit #43: Add scan size that will be used to MultiAcquire UI

Brow71189, 2020-10-15:


cmeyer, 2020-10-15:
Will this just be a display or an editor? Is there a common way we could do this for both SI and MultiAcquire?

Brow71189, 2020-10-16:
For now I would just add it as a display just to make it clear for the user what he gets. 
In the long run it would be nice to have a common way for SI and MultiAcquire. Also MultiAcquire should be able to use a subscan region or a line scan graphic just like regular SI. For that I think it would also be nice to have a common way of defining this. Right now all the code for these things is in a module in the ``nionswift_plugin`` namespace and is closely tied to the SI tool (see also #25).

cmeyer, 2025-03-20:
See
- https://github.com/nion-software/nion-internal/issues/86 (private issue for new API)
- https://github.com/nion-software/nionswift-instrumentation-kit/issues/165

closed 2025-03-20 by cmeyer