nion-software/nionswift-instrumentation-kit #44: Add the SI startup overhead to the MultiAcquire time estimate

Brow71189, 2020-10-15:
For now this can only be some estimate. If we want to be more accurate cameras need to allow querying their overhead.

cmeyer, 2020-10-15:
This needs to be done for regular SI too.

Brow71189, 2020-10-16:
Would be nice to have for regular SI, but it is not as critical as for MultiAcquire. There you even have the overhead from starting a new SI + finishing up (like save everthying to disk etc) and of course the overhead adds up over time. If you do small and fast scans, say a 0.5 s SI and you have 1 s before and 1 s after the SI for preparation and finishing up, after 10 SIs you will have spent 5 s on acquisition (which would also be the displayed time estimate) but 20 s in addition for overhead. For a single SI you might not even notice that, but for multiple SIs this is very noticeable.

cmeyer, 2025-03-20:
Merged with #165

closed 2025-03-20 by cmeyer