nion-software/nionswift-instrumentation-kit #54: Improve error detection during synchronized acquisition

cmeyer, 2020-12-02:
If synchronized acquisition fails, the timeout is long and the user is not given much indication that it has failed other than a Cancel button.

Ideally, the scan should throw an exception if it fails to start scanning within a reasonable amount of time.

Additionally, the overall timeout should be estimated more accurately.

And finally, if these conditions occur, the user should be clearly notified of the issue.

jamesrussell216, 2025-01-06:
@Ion-e has this ticket got everything needed in it from #204? If so can you close the other one please?

Ion-e, 2025-01-06:
> If synchronized acquisition fails, the timeout is long and the user is not given much indication that it has failed other than a Cancel button.
The "Cancel" button is not necessarily a confirmation that it failed, but that it's attempting to run the acquisition.

> And finally, if these conditions occur, the user should be clearly notified of the issue.
Just to clarify, I would expect an Acquisition to notify the user even if it doesn't fulfil all the aforementioned criteria (i.e: the error triggered is different)

Aditionally:
The Cancel button should be responsive and cancel an acquisition as soon as possible.

Also I would add that on testing this fix, it should be tested in different types of Acquisition, not only Scan Synchronised.

Ion-e, 2025-01-27:
On test, check the following case: #204