nion-software/nionswift-instrumentation-kit #148: Arbitrary limit on scan size in ScanFrameDataStream

Brow71189, 2023-02-09:
https://github.com/nion-software/nionswift-instrumentation-kit/blob/8299a94d9862143f932f04270111afa6c2bf1209/nion/instrumentation/scan_base.py#L1989

This puts a hardcoded limit 0f 2048 x 2048 pixels on the scan size in a data stream. A customer request was to be able to acquire 4096 x 4096 pixel scan sequences with the Acquisition panel.
Is there a reason we cannot allow larger scans than what is hardcoded here? Should we increase the limit if possible?

cmeyer, 2023-02-09:
I can't think of any reason for this restriction - but it's strange how I added code to adjust the height based on the restriction. However, before changing, we should test it as there may be a low level limitation. Also, I don't think we should remove it completely: we should leave an assert that the overall size is reasonable, perhaps smaller than 32768 x 32768? this ensures that passing an infinity doesn't somehow try to allocate a huge amount of memory.

cmeyer, 2023-06-21:
b9f267f7262a545e3827d973a852391e9ffd0e48

closed 2023-06-21 by cmeyer