nion-software/nionswift-instrumentation-kit #182: Support mask variations suitable for 4D STEM imaging

cmeyer, 2024-04-15:
Copied from a previously internal issue:

AM:

If we want to support things like DPC with the Dectris ELA we need new mask types to allow the user to create more complex masks. 
See the mask below for example. I used a couple of rectangle and ellipse graphics to create it, but this is not very user friendly.

[image]

Maybe after spend some time with my preliminary mask builder we can decide which building blocks are needed to simplify creating the most common masks.

In the above case maybe something like an "asymmetric wedge mask" would help. 

At the minimum setting up a virtual segmented STEM detector should be as simple as with a hardware segmented detector.
There you typically want to use signals like (Q3 - Q1) or some other linear combination of the signals of the individual segments.

[dpc-stem-imaging-magnetic-structure-application-note.pdf](https://github.com/nion-software/nion-internal/files/5420968/dpc-stem-imaging-magnetic-structure-application-note.pdf)

CM: Another way we might look at this is to figure out what might be parameterized in each of the masks. For instance, we have the origin, the overall angle, the wedge start/end angles, the inner/outer radius. Is this a good parameterization of this particular type of mask?

CM: Also, the dark/light portions are just "the mask", right? There is no difference between those two areas? i.e. they're not 1.0 and -1.0 are they?

AM: Oh no the black area is -1, the grey area is 0 and the white one is +1

Another type of mask we might want is a center-of-mass mask (see Hachtel et al. Sub-Å Electric Field Measurements JASCI 2018).

A center-of-mass mask would have a linear gradient in it, generally with 0 being at the center of the mask/frame. Of course we need to allow combining it with other masks, so that you can for example limit the center-of-mass calculation to a circle at the center of a frame.

MattRoyle, 2026-09-10:
This was discussed in the backlog refinement, the notes are as follows:
I believe this was discussed in SwiftCon and @jamesrussell216 made notes which can be put on the issue.

jamesrussell216, 2026-09-16:
@MattRoyle just looked and that is covered by the parent epic: https://github.com/nion-software/nionswift/issues/1029