nion-software/nionswift-instrumentation-kit #204: When an acquisition fails, it doesn't *tend* to show a warning dialog, instead the "Cancel" button remains iddle

Ion-e, 2024-10-25:
The cancel button "lingers" when an acquisition has failed.
The cancel button is unresponsive.
Ocasionally, when an Acquisition has failed there are no warning dialogs.

I see these as 3 different issues and I would be happy to separate them into 3 different tickets.

Example of acquisition failing "silently" and with a warning dialog, respectively:

[image]

[image]

Ion-e, 2024-10-29:
I believe this issue was originally triggered by something not responding and swift not recieving the expected data.

So I would wonder if there is any type of failsafe for "camera timeout" or other scenearios where the software may remain idle.

Ion-e, 2024-10-31:
Also, please note that when pressing the "Cancel" button wouldn't stop the acquisition.

jamesrussell216, 2025-01-27:
@Ion-e does #54 now cover this one OK?

Ion-e, 2025-01-27:
Happy to make a note on #54 and test this particular case on that ticket once it's been resolved. Thanks!

closed 2025-03-20 by cmeyer