nion-software/nionswift-instrumentation-kit #205: Acquisition failed: No simulator thread when using a shortEELS exposure time on the simulator

Ion-e, 2024-10-31:
The title of the ticket refers to the Acquisition error, however I may have encountered different issues.

The issue comes from attempting to do a very short exposure time (I attempted 0.01ms to which it defaulted back to 0.0ms).

After the issue was observed, it seems the simulator broke down (Even putting the exposure back to over 1ms, every attempt to play the camera will throw the error.

Also, when introducing exposures under 0.0xms, the exposure seems to reset back to 0.0ms.

[image]

cmeyer, 2025-03-20:
@Ion-e Can you revisit this? Is this a bug with real acquisition or just the simulator?

Ion-e, 2025-03-21:
To test on AppDev.

Ion-e, 2025-03-27:
Issue persists on simulator.