nion-software/nionswift-instrumentation-kit #218: Acquisition breaks when frame count is only incremented by 1 in an iteration

Brow71189, 2025-01-30:
This is the problematic line:

https://github.com/nion-software/nionswift-instrumentation-kit/blob/d44728a38d468b233bc7b011705875aff1171c5b/nion/instrumentation/Acquisition.py#L1809

if `count == 1`, the system assumes that it is only acquiring a single frame, and then tries to reshape the data to `data.shape[1:]`. But also a sequence acquisition that only incremented the frame count by 1 in an iteration will lead to `count == 1`, resulting in an error because you cannot do `data.reshape(data.shape[1:])` unless `data.shape[0] == 1`, but that is not guaranteed here.

I've changed this line to:
```python
if count is None or (count == 1 and data_stream_event.source_data.shape[0] == 1):
```
which seems to work so far but might cause other issues. But this definitely needs a more thorough check than what is currently implemented.

cmeyer, 2025-01-30:
This seems to be a dupe of #203 (reported by @Ion-e) and was hopefully fixed yesterday by d44728a38d468b233bc7b011705875aff1171c5b. Can you confirm?

Ion-e, 2025-01-31:
Unsure. I can't manage to reproduce this bug, is there any way I can get some Steps To Reproduce? (STRs) @Brow71189?

Ion-e, 2025-02-19:
Issue verified as fixed. Thanks!