nion-software/nionswift-instrumentation-kit #221: Acquire button has seemingly arbitrary states

Ion-e, 2025-01-31:
The "Aquire" button in the Acquisition panel seems to sometimes be Enable when it's not functional, and viceversa:

- Former example: Untick the HAADF and MAADF tick boxes in the Scan Scan Control > Button is enabled but non functional when on "Scan" mode.
- Latter example: Start Swift > Attempt to do a EELS Scan Synchronized < Button is disabled, however there shouldn't be anything stopping this acquisition as it would be ready to go.

cmeyer, 2025-03-14:
The Acquire button is only active if a scan context has been established by scanning at least once.

Ion-e, 2025-03-25:
Yes, is that the behaviour a user would expect?

jamesrussell216, 2025-03-27:
I've discussed this one previously with @cmeyer but I think it might be worth us talking about it as a wider group. A lot of the reasons for the Acquire button being disabled without a context scan being taken comes back to workflow. For example is there a sensible use case for a user to do an EELS acquisition, without a context HAADF scan? Probably not, with the possible exception of Swift having been restarted due to a crash/failure. Having the Acquire button disabled without a context scan being before is an attempt to steer the user back into a sensible workflow, but it's not exactly obvious.

As I say I discussed this with @cmeyer and he suggested that perhaps we've had this the wrong historically and we should just enable it. I'm inclined to agree with that short-term, though I think when we do a wider look at SuperScan we should think more in terms of workflow.

jamesrussell216, 2025-05-14:
So I believe this isn't a thing anymore after the changes to the Acquire button effectively fixed this.

@Ion-e can this be closed now?

Ion-e, 2025-06-10:
> Former example: Untick the HAADF and MAADF tick boxes in the Scan Scan Control > Button is enabled but non functional when on "Scan" mode.
This is still an issue. You could argue that the button should be disabled, we have the ability to do that already and it would indicate to the user that they are about to do something silly.

The second issue is indeed, already fixed.

Ion-e, 2025-06-13:
It's advisable to restest the current behaviour from Swift startup as the behaviour may have changed with recen changes in the software. @jamesrussell216