nion-software/nionswift-instrumentation-kit #257: Spectrum Imaging throwing a timeout error with usim

jamesrussell216, 2025-05-06:
Testing using usim with the shown parameters, seems to be throwing a timeout error not sure if usim specific though...

[image]

cmeyer, 2025-05-06:
Fails with the equivalent (except 50ms instead of 100ms) acquisition panel. Works with 10ms.

[image]

cmeyer, 2025-05-07:
The problem is progress is not reported correctly in some cases, but is used to determine if the acquisition is proceeding.

cmeyer, 2025-05-07:
This is fixed in 2cf43b72bd0fb2bc5bb7b32959fd209575c8641e, without PR since it is one line.

closed 2025-05-07 by cmeyer