nion-software/nionswift-instrumentation-kit #270: STEM controller should be aware of beam path for camera insert/retract controls

cmeyer, 2025-06-03:


jamesrussell216, 2025-06-11:
Yes @Ion-e

cmeyer, 2025-06-11:
This would be general behavior for the instrumentation kit and camera devices within the instrumentation kit.

cmeyer, 2025-12-16:
@Brow71189 Do you have any ideas for this?

For instance, maybe a method like "is detector on beam path"? Or "detector order"? Something else?

Are hard coded heuristics a viable way to do things initially? Or is this different from instrument to instrument?

Steps:
1. Decide on API
2. Decide on concrete methods are hard-coded for our specific instrument or needs configuration file.
3. Implement API for our instruments.

Brow71189, 2025-12-16:
I think we need to define the usecase for this first.
Is STEM controller expected to insert/retract detectors automatically when a user enters a certain mode of operation (somewhat like the radio buttons in the AS tak bar)? Or does STEM controller just query AS2 for the mode and then sets a flag for each detector whether it can be used in that mode? Do we want to include STEM detectors? Except for the HAADF, they also block the beam for any detector setting beind them.